StratifyLabs/StratifyOS #8: Low power mode support?

ramonschepers, 2016-05-26:
Does stratify make use of the mcu's low power mode(s) when idling, is there support to add this, as its mcu specific?

Might get to port stratify OS to a smartwatch project I'm working on

tyler-gilbert, 2016-05-26:
Yes. It uses a powerdown() and hibernate() function.  powerdown() is rarely used because it resets the MCU.  hibernate() can wake up from the RTC or other external interrupt.

The functions are documented here:
http://stratifylabs.co/StratifyOS/html/group___s_t_f_y.html

We are developing a smartwatch oriented board based on the LPC54xx and low energy bluetooth.  You can also base your design on the LPC17xx or LPC407x/8x both of which are already fully supported.

ramonschepers, 2016-05-26:
That's good to hear, though I wonder why nxp micros are the base, I personally prefer st micros. 

Also, I can't find any contact details or a Twitter account to interact with?

ramonschepers, 2016-05-26:
Ah, I just looked at the data sheet of the lpc540xx and it has 2 CPU cores, perhaps one could act as a GPU?

tyler-gilbert, 2016-05-26:
The lpc54xx optionally has 2 cores.  If you need powerful graphics, you should consider the LPC4088 with the built-in LCD controller.  If you are using a SPI based display, the LPC54xx should work.

Stratify OS has deep hardware integration.  So we try to provide fully-featured drivers for all the available hardware so our users only have to write their application and not worry too much about writing MCU driver.  This means it is a huge effort to port to another mfg.  That being said, the LPC and STM32 lines of chips are pretty comparable on both cost and features.

ramonschepers, 2016-05-28:
I simply need a spi bus for the screen, and another question: how easy would it be to integrate a single framebuffer of 128*128px 16bit (32k RAM)

And to come back to my original question: is the OS event driven and each moment inactive means powerdown() ?

tyler-gilbert, 2016-05-28:
It isn't too difficult to implement a frame buffer using the SPI bus and a timer.

OS is not event driven.  You can implement your own event driven app framework though.

tyler-gilbert, 2016-05-31:
I am going to close this issue if you don't have any more question regarding low power mode.  Sound good?

closed 2016-05-31 by ramonschepers