StratifyLabs/StratifyOS #233: sos-toolchain.cmake doesn't respect custom SDK install directories

syntroniks, 2020-08-25:
Using mingw on windows,

```
me@PF1AJJFF MINGW64 ~
$ echo $SOS_SDK_PATH
/c/Users/me/StratifyLabs-SDK
```

```
me@PF1AJJFF MINGW64 ~/Documents/path (branch)
$ sl sdk.update
# sdk.update

## output
- tasks:
   - StratifyOS:
      - pull: StratifySDK/StratifyOS
### external.begin

Already up to date.

### external.end

      - compile:
         arch: arm
         cmake options: -DSOS_SKIP_CMAKE=OFF -DBUILD_ALL=ON -DSOS_ARCH_ARM_ALL=OFF -DSOS_ARCH_ARM_V7EM_F5DH=ON
         make options: -j8
         clean: false
         generator: <default>
         build: install
         reconfigure: false
      - StratifySDK/StratifyOS:
         configure:
### external.begin

-- SOS BUILD CONFIG PROVIDED: arm
-- Windows provided toolchain directoryC:/StratifyLabs-SDK/Tools/gcc
-- User provided toolchain directoryarm-none-eabi
-- GCC Toolchain is: C:/StratifyLabs-SDK/Tools/gcc/arm-none-eabi
-- The CXX compiler identification is unknown
-- The C compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: C:/StratifyLabs-SDK/Tools/gcc/bin/arm-none-eabi-g++.exe
CMake Error at CMakeLists.txt:15 (project):
  The CMAKE_CXX_COMPILER:

    C:/StratifyLabs-SDK/Tools/gcc/bin/arm-none-eabi-g++.exe

  is not a full path to an existing compiler tool.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:15 (project):
  The CMAKE_C_COMPILER:

    C:/StratifyLabs-SDK/Tools/gcc/bin/arm-none-eabi-gcc.exe

  is not a full path to an existing compiler tool.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:15 (project):
  The CMAKE_ASM_COMPILER:

    C:/StratifyLabs-SDK/Tools/gcc/bin/arm-none-eabi-g++.exe

  is not a full path to an existing compiler tool.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!

### external.end

         error: failed to execute cmake
   - result: fail

```

I suspect the issue may be related to https://github.com/StratifyLabs/StratifyOS/blob/da36672078b9497b5dc297c84967cbcd3d02e52f/cmake/sos-toolchain.cmake#L21

syntroniks, 2020-08-25:
Even if it was hitting the correct directory, it looks like `sl sdk.install` didn't do anything (I'm obtaining this copy from the new link, not `..../slwin`)

syntroniks, 2020-08-25:
I have issues with sl version 0.67
```
$ sl --version
# about

## output
- publisher: Stratify Labs, Inc
- version: 0.67
- gitHash: 49c800d
- apiVersion: 3.20.3
- apiGitHash: 97be0da
- sdk: C:/StratifyLabs-SDK
- system: windows_x86_64
```
But not with version 0.52
```
$ sl.exe --version
# about

## output
- publisher: Stratify Labs, Inc
- version: 0.52
- gitHash: 4f0a425
- apiVersion: 3.15.0
- apiGitHash: 4abb3ce
# update
latest: 0.70
installed: 0.52
- downloading: ##################################################
update: true
skip: `sl settings.set:key=skipUpdate,value=0.70,add,g`
```

tyler-gilbert, 2020-08-25:
OK. I am looking into this one and the linux one.